adafruit/Adafruit_SleepyDog #4: Time keeping in powerdown mode

shirish47, 2015-12-28:
Hi, 
I have this need of time keeping in my arduino. although its not necessary to have a perfect timing but if it was good enough for mins resolution it would be good. I wanted to use Time library and also lowpower library, in powerdown mode from lowpower library.. timer0 is shutdown too .and thus millis() fail and also Time library.. so I wanted to have timekeeping and thus I though time keeping with watchdog timer..in low power there is a mode to wake controller ever 1,2,4,8 secs.. I can adjust the timer library seconds at every wakeup and check if the wake up is not from Watchdog it will not update seconds .. this is good for 1 second sleep ... I hope some one can implement this in any of these libraries ..this way we can get a rude low power with time keeping library..but this should not be used for realtime things ...what do you think of this?

tcustodio, 2016-05-07:
Hi, shirish47. I ran into a similar problem a few days ago and right now what I'm doing, using both the Time library and the SleepyDog is this:
`adjustTime(Watchdog.sleep()/1000);`
This line will take the time that the arduino slept (e.x.: 8000ms) divide by 1000 to convert from miliseconds to seconds and offset the internal timekeeping of the Time library.
Not yet a complete solution but better than none. It seems that in the long run the clock drifts a few seconds using this method. I hope I have helped in any way

shirish47, 2016-05-12:
ya it works fine in my project. few seconds is not a problem. and later we can update it with NTP server once a day. RTC is also possible solution for this. I have not tried it but Atmega328 has RTC and I think that would have low power consumption than WDT, I suppose. not sure would try someday and post it.