adafruit/Adafruit_SleepyDog #7: Watchdog.sleep() and Feather Lora M0

lbovay, 2017-02-19:
Hi,
did anyone get Watchdog.sleep(xxxx) to work with the Feather Lora M0 ?
Mine goes to sleep but never wakes up :(

I use the example code here:
`// Adafruit Watchdog Library Sleep Example
//
// Simple example of how to do low power sleep with the watchdog timer.
//
// Author: Tony DiCola

#include <Adafruit_SleepyDog.h>

void setup() {
  Serial.begin(115200);
  while (!Serial) ; // wait for Arduino Serial Monitor (native USB boards)
  Serial.println("Adafruit Watchdog Library Sleep Demo!");
  Serial.println();
}

void loop() {
  Serial.println("Going to sleep in one second...");
  delay(1000);
  
  // To enter low power sleep mode call Watchdog.sleep() like below
  // and the watchdog will allow low power sleep for as long as possible.
  // The actual amount of time spent in sleep will be returned (in 
  // milliseconds).
  int sleepMS = Watchdog.sleep();

  // Alternatively you can provide a millisecond value to specify
  // how long you'd like the chip to sleep, but the hardware only
  // supports a limited range of values so the actual sleep time might
  // be smaller.  The time spent in sleep will be returned (in
  // milliseconds).
  // int sleepMS = Watchdog.sleep(1000);  // Sleep for up to 1 second.

  Serial.print("I'm awake now!  I slept for ");
  Serial.print(sleepMS, DEC);
  Serial.println(" milliseconds.");
  Serial.println();
}`

Cheers,

PaintYourDragon, 2017-02-19:
Code's there and works, but just noticed I didn't bump the version number, so Arduino IDE might not pick up the latest. I'll do that in a couple minutes here (to version 1.1.1), but may take some time to filter up...can either wait a little while, or download and install ZIP if needed immediately.

closed 2017-02-19 by PaintYourDragon

lbovay, 2017-02-19:
Thank you for your answer,

Still no luck with the Sleep example. I'm running Arduino 1.8.1, I installed this library in V1.1.1 aswell as this dependancy: https://github.com/adafruit/Adafruit_ASFcore

Code is building without issues, console output is:
`Adafruit Watchdog Library Sleep Demo!

Going to sleep in one second...
`

PaintYourDragon, 2017-02-19:
Not sure what to suggest. I'm running same version of IDE & libraries. Slightly different board (RFM69 instead of LoRa), but that shouldn't make a difference, all the M0 boards should run that code identically. MAYBE a compiler difference (I'm on Mac) but haven't seen that sort of thing in years.

scott2b, 2018-04-10:
Re: the LoRa boards. If you are using RadioHead, iirc there is a conflict between the radio driver and Sleepy Dog. I believe it was the case when I tried it that they were competing for the same interrupt or something like that. Otherwise, if not using the RadioHead driver for the LoRa, Sleepy Dog seems to work fine.

One caveat. I might not look like it is waking up if you are depending on Serial output. It will lose the serial connection when it goes to sleep. Also, re-loading can be quite finicky if you have a sleep in your loop. I find that I have to kick the Feather by pushing the reset button 2x in order to load new code.

huzarif97, 2018-05-30:
I am currently trying to use both RadioHead and Sleepy Dog and am experiencing the issue scott2b is describing; I found a quick fix where I comment out the area of tc_interrupts.c that creates the TCx_Handler for each TC, but I would like to find a better way that salvages all of the TC's in case I need them for the future.  Thanks

Kuurusch, 2018-06-03:
Hey, is this issue fixed for the LoRa-modul on the M0?

Kuurusch, 2018-06-16:
Had someone luck with the radioHead and SleepyDog?

loicgillioz, 2018-07-06:
Hi, i've effectively found that going into sleep mode with the LoRa module used makes it crash. I will keep you updated if I find a fix.

Kuurusch, 2018-08-04:
@loicgillioz Hey, thanks a lot, yea would be cool. Otherwise the M0 needs way too much energy :/

nyfelix, 2020-04-29:
Thank you guys, after testing for almost an hour I realised, that after sleep the Serial connection was broken... but it works all fine.