adafruit/Adafruit_SleepyDog #13: Conflict with radiohead library on Adafruit Feather M0

2lostkiwis, 2018-06-12:
- Arduino board:  Adafruit Feather M0 with RFM95 LoRa Radio
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.5

Library conflicts with radiohead. The the above Adafruit board, it's quite likely to want to use these two libraries together - but both appear to use the TC3 interrupt handeller. Giving the following error when you try to use them together:

`libraries\Adafruit_ASFcore\tc_interrupt.c.o: In function 'TC3_Handler':
C:\Users\ibarnes\Dropbox\Arduino\libraries\Adafruit_ASFcore/tc_interrupt.c:140: multiple definition of 'TC3_Handler'
libraries\RadioHead\RH_ASK.cpp.o:C:\Users\ibarnes\Dropbox\Arduino\libraries\RadioHead/RH_ASK.cpp:611: first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit Feather M0.`

ladyada, 2018-06-13:
yeah what we need to do is de-ASF this library, no ETA as to when we'll get to it!

Kuurusch, 2018-06-16:
Would be cool to see a fix for this :)

Pastant, 2018-06-16:
https://learn.adafruit.com/adafruit-feather-m0-radio-with-lora-radio-module?view=all

You say: "~300uA during full sleep" ... How do I put this thing to full sleep? The SleepyDog library apparently doesn't work. 

I reported this error a year ago:

https://forums.adafruit.com/viewtopic.php?f=57&t=121415

StefanMeGit, 2018-11-16:
Any progress here? I need this library urgently together with RadioHead

ladyada, 2018-11-16:
we'll take a look - no promises whatsoever on when it'll be done :)

Kuurusch, 2018-12-26:
Thanks a lot, would be really cool!!!!

Pastant, 2018-12-26:
Any eta?

ladyada, 2018-12-26:
no eta, we did de-ASF it, but have not tested with radiohead

frax84, 2019-02-12:
I have no problem in compiling the code with both libraries on Feather 32u4 but i have some conflict too. In particular, when i use "Watchdog.sleep()" i get "waitPacketSent()" of RadioHead library (v.1.89) hangs indefinitely. Is this related to this issue?

jgovindu, 2019-03-22:
Any update on radiohead libary support for the m4 board or any boards that use the SAMD51 processor?

ladyada, 2019-03-22:
radiohead should work just fine with radiohead: https://github.com/adafruit/RadioHead

Kuurusch, 2019-08-29:
So, the original problem is solved?

Pastant, 2019-08-29:
The problem is not solved ... seems as if there will be no fix. I reported the bug 2 years ago!

ladyada, 2019-08-29:
hiya the file has been deleted, what error do you see now?

Kuurusch, 2019-08-30:
So I can confirm, the bug is solved! I've updated the IDE and all libraries and now it works! Great job, thanks a lot! The only thing is, that the maximal time to put the M0 into deep sleep is somehow quite limited. I've got no more than 20 seconds! Is that true? So at the moment I check, how long the M0 was sleeping and put it immediatly to sleep again, when my total sleep-time is not already reached.

Pastant, 2019-08-30:
Same here. I had to delete all the old librarys. At first I just renamed the folders. Thanks!

closed 2019-08-30 by ladyada