adafruit/Adafruit_SleepyDog #18: possible update for SAMD51

ladyada, 2019-01-17:
https://forums.adafruit.com/viewtopic.php?f=62&t=145747

MartinL1, 2019-03-02:
I've been testing the SAMD51 in STANDBY mode, on both the Metro M4 and Itsy Bitsy M4, but neither enter deep sleep after receiving the __WFI() command. However, the SAMD21 (on my Arduino Zero) functions correctly.

After some research I found that the USB peripheral module is set to run on standby (RUNSTDBY) on both the Arduino Zero and the M4 boards. It appears as though on the SAMD51 boards that interrupts from this peripheral are keeping the processor awake. If I add these lines to the beginning of my sketch, to shutdown the USB, the SAMD51 goes into deep sleep and both external and watchdog interrupts wake it up correctly:

```
USB->HOST.CTRLA.bit.ENABLE = 0;                   // Shutdown the USB peripheral
while(USB->HOST.SYNCBUSY.bit.ENABLE);             // Wait for synchronization
```
The question remains however, as to why this occurs on the SAMD51 boards and not on the SAMD21 Arduino Zero.

HowardStark, 2019-03-05:
@MartinL1 Thank you **so much**. I've been working on this for the past week and was just about to essentially rebuild the SAMD51 Arduino garbage all from scratch because I was at such a loss. Found your post which linked here. 

If anyone can't get their SAMD51 device to sleep, this very well might be it.

HowardStark, 2019-03-05:
Here's the line that does it for anyone that's curious like me:
https://github.com/adafruit/ArduinoCore-samd/blob/samd51/bootloaders/metroM4/board_driver_usb.c#L130

ladyada, 2019-03-06:
thanks - plesae let us know if it works and we could add it to the sleepydog library (it'd shut down USB before sleeping)

HowardStark, 2019-03-06:
@ladyada Sorry I didn't convey that I had already tested earlier. I can confirm that on an ItsyBitsy M4 (SAMD51G19A) this fix works and the device is sleeping with all the expected behavior.

MartinL1, 2019-03-06:
@HowardStark 

Glad to hear that you also got your SAMD51 to sleep as well. 

I don't think that the issue is with the SAMD51's bootloader code though.

I'm able to put the SAMD51 to sleep with __WFI() by simply commenting out the line 375 in the "USBCore.cpp" file in the  .../USB/... directory of the M4 boards' core code:

```
// usbd.runInStandby();
```
...this function just sets the RUNSTDBY bit in the USB module's CTRLA register.

ladyada, 2019-03-06:
ok so the only change is to comment that out in the samd BSP repo, and then this library is all good? if so can someone please submit a PR there - we'll pick it up for testing into the `master` branch and if good it will go into the next release ! :)

LarsHolstHansen, 2019-04-03:
When is this next release planned for?

ladyada, 2019-04-03:
next week or two? we do them whenever we have a couple fixes to distribute

ladyada, 2019-10-05:
fixed via https://github.com/adafruit/Adafruit_SleepyDog/pull/19 and this causes bootloader problems so rejected

closed 2019-10-05 by ladyada