adafruit/Adafruit_SleepyDog #20: BasicUsage example sketch fails on Metro M4 express

dcliftreaves, 2019-06-02:
I tested the BasicUsage sketch on a Metro M4 express. It arrived from adafruit with a stock bootloader of 2.0.0. The sketch failed by getting to the main loop without resetting.

I upgraded the bootloader to 3.6.0.
Tried the sketch again and it still displays the main loop message indicating an error.

I even tried adding this code to the top of the setup function:
USBDevice.standby();
(Strangely, this calls "noRunInStandby" behind the scenes.)
Based on the other issue described in this repo regarding the SAMD51. #18 
(Although, it is possible I interpreted the proposed root cause incorrectly.)

dcliftreaves, 2019-06-02:
I just noticed that I used the last release through the arduino interface and not the latest commit on this repo.
I tried that and the code doesn't print anything in the main loop, like it did before, but it also doesn't appear to reset the device. (No additional text is printed through the usb interface anyway.)

dcliftreaves, 2019-06-02:
I see what is going on. When the device is reset the Arduino serial terminal doesn't handle that or display that change in status. The sketch technically does work now but because the device itself is hosting the USB CDC connection when it is reset the serial terminal disappears. You would have to create a system for auto-reconnecting to it to make it work using this method.

Well, at least the BasicUsage sketch appears to work based on the latest repo but not the latest release.

dcliftreaves, 2019-06-02:
I am not closing this because I believe simply an updated release with the latest code is required to close it. Feel free to override that decision!

ladyada, 2019-10-05:
hiya tested latest code and it works so closing

closed 2019-10-05 by ladyada