adafruit/Adafruit_SleepyDog #22: When trying to connect to GSM with arduino GSM1400. Watchdog timer resets the board

dimilio81, 2020-06-12:
I am trying to use the sleepydog library in a project that i am building that uses the GSM1400 arduino Board. The library works fine always until you try to connect with the arduino library to GSM network. If you disable the watchdog, the GSM connection is successful and works normally. And when i say it resets the board, the reset is immediate, it does not occur after the time that the watchdog is setup passes. Any ideas?

SWoto, 2020-07-22:
I've a problem like yours.

I'm using an XBee with a SAMD51 board with the Adafruit_SleepyDog library. The code runs smoothly if I remove the line `uint32_t sleepMS = sleepForSeconds(SLEEP_TIME);`, it spams my XCTU with messages. However, after sleeping the SAMD51, I can't get it to work. I even tried checking `Serial2.avaiable()`, but it didn't work either. Even though it doesn't uses the USD Serial, I've also tried `USD.attach()`.

Rock-N-RollaMann, 2020-07-22:
Hi, I think it’s better to use the RTC Zero library. I use it instead of the sleepy dog library and works much better.

SWoto, 2020-07-23:
I've tested using an LCD to display (because the USB doesn't work after waking it up) that uses SPI communication and it worked.

SWoto, 2020-07-23:
@MartinL1, did u managed to make the USB/Serials work after waking the board up?

Rock-N-RollaMann, 2020-07-24:
I don't know who MartinL1 is in this thread, but this code works for me:

`USBDevice.detach();
// Reinitialize USB for debugging
USBDevice.init();
USBDevice.attach();`
USBDevice.detach();
// Reinitialize USB for debugging
USBDevice.init();
USBDevice.attach();

SWoto, 2020-07-29:
With this code, even though I can still see the USB, I can't print anything. However the code is still working, except for the Serial, the LED keep blinking in the correct time.

[image]
 
```
  digitalWrite(LED_BUILTIN, LOW); // Show we're asleep
  sleepMS = Watchdog.sleep(4000);
  digitalWrite(LED_BUILTIN, HIGH); // Show we're awake again
  USBDevice.attach();
  delay(2500);
```

With `USBDevice` parts it doesn't work a second time. The led goes on, then off when sleeping, on again when it awakes than nothing more, it gets stuck somewhere.
```
  USBDevice.detach();
  digitalWrite(LED_BUILTIN, LOW); // Show we're asleep
  sleepMS = Watchdog.sleep(4000);
  digitalWrite(LED_BUILTIN, HIGH); // Show we're awake again
  USBDevice.init();
  USBDevice.attach();
```
[image]

SWoto, 2020-07-29:
I'm testing it with a [SparkFun Thing Plus - SAMD51](https://www.sparkfun.com/products/14713).

Full code:
```C
#include <Adafruit_SleepyDog.h>

int times = 0;
int sleepMS = 0;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH); // Show we're awake

  Serial.begin(115200);
  while(!Serial); // wait for Arduino Serial Monitor (native USB boards)
  Serial.println("Adafruit Watchdog Library Sleep Demo!");
  Serial.println();

}

void loop() {
  Serial.println("Going to sleep in one second...");
  delay(3000);

  USBDevice.detach();
  digitalWrite(LED_BUILTIN, LOW); // Show we're asleep
  sleepMS = Watchdog.sleep(4000);
  digitalWrite(LED_BUILTIN, HIGH); // Show we're awake again
  USBDevice.init();
  USBDevice.attach();

  Serial.print("I'm awake now! I slept for ");
  Serial.print(sleepMS, DEC);
  Serial.println(" milliseconds.");
  Serial.println();
  times += 0;
}
```

Rock-N-RollaMann, 2020-07-30:
Try commenting out this line:
` while(!Serial); // wait for Arduino Serial Monitor (native USB boards)`

SWoto, 2020-07-30:
@Rock-N-RollaMann, thanks for your fast reply.

The board would only get stuck there if the board had, somehow, reseted itself. Otherwise it shouldn't go back to the setup...
Anyway, I tried commenting this line and the problem persist. The code is locked somewhere...

SWoto, 2020-07-30:
I commented some parts of the code and also removed the USBDevice.dettach() so I could see the logs.
```
Going to sleep in one second...
sleep: int WatchdogSAMD::sleep(int maxPeriodMS)
    enable: int WatchdogSAMD::enable(int maxPeriodMS, bool isForSleep)
        _initialize_wdt: Before USB->DEVICE.CTRLA.bit.ENABLE = 0;
        _initialize_wdt: After USB->DEVICE.CTRLA.bit.ENABLE = 1;
    enable: #if defined(__SAMD51__)
    enable: After maxPeriodMS and cycles
    enable: #if defined(__SAMD51__)
    enable: if (isForSleep)
sleep: #if defined(__SAMD51__)
sleep: before __DSB() and __WFI()
```

We can see [in the line 247](https://github.com/adafruit/Adafruit_SleepyDog/blob/master/utility/WatchdogSAMD.cpp#L247) that the USB is disabled and than enabled (don't know why it does it just after disabling it).
So I tried to move the line [251 to 253](https://github.com/adafruit/Adafruit_SleepyDog/blob/master/utility/WatchdogSAMD.cpp#L251) to just after the [__WFI();](https://github.com/adafruit/Adafruit_SleepyDog/blob/master/utility/WatchdogSAMD.cpp#L215) command 
```C
__WFI(); // Wait for interrupt (places device in sleep mode)
  Serial.println("after: before __DSB() and __WFI()");
  
  USB->DEVICE.CTRLA.bit.ENABLE = 1;   // Enable the USB peripheral
  Serial.println("sleep: After USB->DEVICE.CTRLA.bit.ENABLE = 1;");
  while (USB->DEVICE.SYNCBUSY.bit.ENABLE)
    ; // Wait for synchronization
```

But the only thing that changed was that I lost the usb connection earlier than before...

mdelain, 2021-11-05:
Coming back to the inital issue:

> I am trying to use the sleepydog library in a project that i am building that uses the GSM1400 arduino Board. The library works fine always until you try to connect with the arduino library to GSM [...]

I also have the same annoying behavior - this library seems to be just incompatible with MKR GSM 1400 as soon as you try to open the GSM connection, and even without putting the board asleep! Or does anyone has a workaround to make it work?

CptHolzschnauz, 2024-01-23:
The issue is IMHO that when the SAMD is interupted by the watchdog, it doesn't read the serial AT console and some bytes get lost. Can lead to a freeze but it's flaky behaviour. The modem is full computer and doesn't know that the SAMD is not listening.