adafruit/Adafruit_SleepyDog #32: Limit on sleep time

pauljurczak, 2021-06-24:
I'm using this library with Adafruit Feather M0 Basic Proto and PlatformIO on VS Code. I noticed that the maximum sleep time is around 30s. Calling `Watchdog.sleep(60000);` or with a larger argument will not increase sleep time.

ladyada, 2021-06-28:
what is the max watchdog time according to the samd21 datasheet?

pauljurczak, 2021-06-28:
I'm not sure. Page 211 of *Atmel-42181G–SAM-D21_Datasheet–09/2015* reads *Selectable time-out periods, from 8 cycles to 16,000 cycles in normal mode or 16 cycles to 32,000 cycles in window mode*. Page 214 states: *In normal-mode operation [..] There are 12 possible WDT time-out (TOWDT) periods, selectable from 8ms to 16s*. So it looks like `Watchdog.sleep()` operates in window mode and provides 16ms to 32s time-outs. It is consistent with my measurements of sleep times as `n/2` when calling  `Watchdog.sleep(n)`, maxing out at around 30s.
.

ladyada, 2021-06-28:
ok!

closed 2021-06-28 by ladyada