adafruit/Adafruit_SleepyDog #37: Support RP2040

brentru, 2022-01-06:
Add support for RP2040 WDT and Deep Sleep modes

While there's no native API calls,
> A complete copy of the Raspberry Pi Pico SDK is included with the Arduino core, and all functions in the core are available inside the standard link libraries.
Usage instructions - https://arduino-pico.readthedocs.io/en/latest/sdk.html

Example SDK code for sleep: https://github.com/raspberrypi/pico-playground/blob/master/sleep/hello_sleep/hello_sleep.c

WDT API: https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__watchdog.html

Phoenix1747, 2022-10-23:
Awesome! Any updates if and when this will be supported?

brentru, 2022-10-24:
@Phoenix1747  Just merged and released! Please give the Arduino Library Manager a few hours to pick up the latest release (https://github.com/adafruit/Adafruit_SleepyDog/releases/tag/1.6.2)

closed 2022-10-24 by brentru

metrafonic, 2023-02-16:
@brentru am I wrong in my understanding that this does not implement any of the deep sleep functionality in `pico/sleep.h` that you linked to as example code? Im not able to achieve any reduction in power usage using this

brentru, 2023-02-16:
The work-related to this issue implements `sleep_ms()`, a lower power (WFE) sleep. This does not implement `sleep_goto_sleep_until()` functionality.

see: https://github.com/adafruit/Adafruit_SleepyDog/pull/41/files