adafruit/Adafruit_SleepyDog #42: Watchdog.Sleep() changes millis() tick rate (Feather M4 Express, SAMD51)

tomjennings, 2023-01-18:
This bug makes Sleepydog Watchdog.sleep() useless. 

If Watchdog.sleep (n) is executed even once, upon return from sleep millis() counts at a much slower rate; the system tick rate is changed. Simple sketch to invoke and see the bug below. Requires an oscilliscope to see. 

Before executing `Watchdog.sleep (1)` pulse period is 100.1 mS. After executing that, pulse period is 114 mS.

Does not occur on AVR boards. I don't have a SAMD21 board to test. 

This code displays the problem.

`#include <Adafruit_SleepyDog.h>

void setup() {

  pinMode (13, OUTPUT);
  pinMode (5, INPUT_PULLUP);
}

void loop() {

  // Period is 100.2 mS at pin 13.
  
  while (1) {
    digitalWrite (13, 1);
    delay (50);
    digitalWrite (13, 0);
    delay (50);
    if (digitalRead (5) == 0) break;
  }

  Watchdog.sleep (1);

  // O'scope indicator that the switch was indeed pressed.
  
  for (int i= 0; i < 10; i++) {
    digitalWrite (13, 1);
    delay (10);
    digitalWrite (13, 0);
    delay (10);
  }

  // Period is 114 mS at pin 13.

  while (1) {
    digitalWrite (13, 1);
    delay (50);
    digitalWrite (13, 0);
    delay (50);
  }
}`

The photos below are before and after the switch press.

BEFORE
[image]

AFTER
[image]

tomjennings, 2023-01-27:
Is anyone working on these things? I don't know the SAMDx1 chips at the register level else I'd poke at it.

I posted this to the Adafruit forum as well, and zero replies. 

I'm personally dead in the water, but I guess few are using the SAMD51 powered Adafruit boards. I see few resources for them.

ruffner, 2023-02-07:
Does this happen when you use ```Watchdog.sleep()``` without the argument?  I will test and report back.

I was planning on using this library, dang, hope this is fixable. I was planning on using FreeRTOS with it, surely there is a way to put this chip to sleep that doesn't involve the watchdog timer. I want indefinite sleep with an interrupt pin wakeup.

ruffner, 2023-02-10:
I am pleased to report back that I have not been able to confirm this as a bug. I used the SleepyDog Sleep example code as a starting point, and separate USB to serial bridge to read debug messages from ```Serial1```. The sketch is attached in a .zip file. The board used to test was a FeatherM4 Express with no modifications.

Adafruit SleepyDog Library version 1.6.3     
adafruit:samd 1.7.10    1.7.10 Adafruit SAMD (32-bits ARM Cortex-M0+ and Cortex-M4) Boards

[Sleeptest_samd51.zip](https://github.com/adafruit/Adafruit_SleepyDog/files/10710823/Sleeptest_samd51.zip)

Using a logic analyzer on pin 4, I generate a 100ms period square wave in ```setup``` before sleeping, and then again after every wake in ```loop```. The first image, showing the period on the left hand wave, is the before sleep trace, and the second image is the wave after waking. As you can see, the periods of the two waves before and after sleep are largely the same, as one would hope to see (I'm surely relieved!)

[image]

[image]

@tomjennings what samd51 board are you using? Can you try my code above and remeasure with the scope?

tomjennings, 2023-02-23:
Hi --

Did you try my test program that induces the problem? I put some effort into reducing to minimum, and not use the USB serial that is disabled by sleep(). The program you provided looks like a barely edited copy of the example program, and worse, refers to non-existent "Serial1".

I'm using an Adafruit Feather M4 Express (SAMD51).

I shortened the test program further. It generates a 100 mS squarewave train using delay(), when pin 5 is momentarily grounded, it executes a single `Watchdog.sleep()` call, then attempts to generate the same 100 mS squarewave but fails, and produces 114 mS period squarewaves.

Mark, since you have an oscilloscope, could you try this exact code and see what happens?

I tried with no argument ("maximum sleep") and 1 (1 millisecond sleep) and both induce the error.

```
#include <Adafruit_SleepyDog.h>

void setup() {

  pinMode (13, OUTPUT);
  pinMode (5, INPUT_PULLUP);
}

void loop() {

  // Period is 100.2 mS at pin 13.
  while (1) {
    digitalWrite (13, 1);
    delay (50);
    digitalWrite (13, 0);
    delay (50);
    if (digitalRead (5) == 0) break;
  }

  Watchdog.sleep ();  // tested with no arg ("max sleep") and 1 (1 mS).

  // Period is 114 mS at pin 13.
  while (1) {
    digitalWrite (13, 1);
    delay (50);
    digitalWrite (13, 0);
    delay (50);
  }
}
```

ruffner, 2023-02-27:
Hi @tomjenning. I am curious what version of the Adafruit SAMD board package and SleepyDog library you are using. As I mentioned before, my versions are:

Adafruit SleepyDog Library version 1.6.3
adafruit:samd 1.7.10 1.7.10 Adafruit SAMD (32-bits ARM Cortex-M0+ and Cortex-M4) Boards

I went back to the scope with a bare Feather M4 and copied your above code into a new sketch and uploaded. My testing showed that after the processor wakes from sleep, it still generates a waveform with a 100ms period. 

As evidence, here is a short video: https://www.youtube.com/shorts/lamQaJn2ijo

Also, FYI, Serial1 is a real thing and refers to the SERCOM5 hardware serial port on the Feather M4 using pins 0 and 1. 

I propose that this issue be closed.

tomjennings, 2023-02-27:
Hi Mark --

Thanks for looking at this for me, I do greatly appreciate it.

I am running
Adafruit SleepyDog Library version 1.6.3
Adafruit SAMD 1.7.11 via Boards Manager
on IDE 1.8

Are you using 1.8, 2.0, or ? I can't see it mattering, but results here are odd...

I've tested on multiple Feather M4 Express boards, SAMD51J19 processor, the bare board dangling on the end of the USB cable, no shields or external components. My boards are a couple years old; do you think there's SAMD51 chip differences, yours to mine?

I watched your video, yup, different result. I only asked if you'd run my test because you hadn't explicitly said you had.

Here's a video of my test, just repeated: https://www.sr-ix.com/temp/PXL_20230227_190340919.TS.mov

I tested on two boards with identical results.

I know about SERCOM generated serial devices, I've configured them on Grand Central. If Serial1 is a synonym for the USB serial I'd forgot. In any case I avoided it's use for this test case.

I think it's premature to close this. 

This is the exact code I used. I dropped the /O'scope indicator/ loop as superfluous. Watchdog.sleep (1).

```
#include <Adafruit_SleepyDog.h>

void setup() {

  pinMode (13, OUTPUT);
  pinMode (5, INPUT_PULLUP);
}

void loop() {

  // Period is 100.2 mS at pin 13.
  
  while (1) {
    digitalWrite (13, 1);
    delay (50);
    digitalWrite (13, 0);
    delay (50);
    if (digitalRead (5) == 0) break;
  }

  Watchdog.sleep (1);  // tested with no arg ("max sleep") and 1 (1 mS).

  // Period is 114 mS at pin 13.

  while (1) {
    digitalWrite (13, 1);
    delay (50);
    digitalWrite (13, 0);
    delay (50);
  }
}
```

tomjennings, 2023-02-28:
For the record I did test with `Watchdog.sleep()`, got the expected ~ 16 sec sleep, but it also munged the tick rate, and since 1mS << 16000mS I stuck with that.

ruffner, 2023-03-02:
Any joy downgrading to 1.7.10?

tomjennings, 2023-03-10:
I "reverted" by deleting Adafruit SAMD support, quit IDE, restarted, installed Arduino SAMD 1.7.10 vi board manager. 

Same results: 100 mS before, 114 mS after `Watchdog.sleep (1)`.

Not sure how to proceed. My short term solution is a bigger battery and solar panel. Not a solution I like.

What difference between you and me could there me? I'll again completely delete all Arduino everything, reboot, install from scratch, etc. But I feel like I'm lighting candles in a church.

[image]

ruffner, 2023-03-14:
I can understand your frustration. Maybe in the meantime try the ItsyBitsy M4 boards to try it with: https://learn.adafruit.com/introducing-adafruit-itsybitsy-m4. 

Does anyone at Adafruit have a 2 year old feather board to try this test on? @brentru what do you think about this discussion?

tomjennings, 2023-03-30:
Thanks Matt. I'm conveniently (sic) side-stepping the problem by ditching Adafruit's platforms. Their bootloaders are awful for Arduino use (require very fiddly RESET double-click for each upload, painful for physically embedded boards), I wish them luck with their CircuitPython thing. On top of that SAMD51 chips have been backordered for months. 

I'm switching to an I2S system, for playback and negating the need for the big processor for native MP3, done and done.

I cross-posted this, politely, on Adafruit forum and not one reply.  So I guess I'll close the issue with this comment.

Thank you Matt for your help and your time. Puzzling, isn't it!

closed 2023-03-30 by tomjennings